FourThievesVinegar/solderless-microlab #7: Additional Instructions?

samhavens, 2020-04-15:
Some additional instructions that may be helpful... Might be worth adding

* [Emulating a Raspberry Pi with Virtualbox](https://raspberrytips.com/run-raspberry-in-virtual-machine/)
* Hardware shopping list
  * rasberry pi 3 - (link)
  * [DC, stepper motor control + limit switch](https://www.aliexpress.com/item/Free-shipping-cnc-shield-v3-engraving-machine-3D-Printer-4pcs-A4988-driver-expansion-board-for-Arduino/32808916949.html)
  * [Interface + buttons](https://www.aliexpress.com/item/Free-shipping-LCD-module-Pi-TFT-3-5-inch-320-480-Touchscreen-Display-Module-TFT-for/32587995145.html)
  * [Relay board](https://www.aliexpress.com/item/1PCS-Raspberry-Pi-Power-Relay-Board-Expansion-Module-Shield-Supports-RPi-A-B-2-B-3/32961638909.html)
  * [Temperature](https://aliexpress.com/item/32827261401.html)
  * [some stepper motors with female headers](https://www.aliexpress.com/item/32853499120.html)

samhavens, 2020-04-15:
Hmm, it doesn't seem like emulating the Pi works, `RPi.GPIO` still seems to be aware it isn't actually running on a Pi.

I will dig out my Pi knockoff and see if that works. Otherwise, I will wait for the Pi I ordered to arrive

SpikeNebula, 2020-04-18:
Theo added the parts list in eccdbb76175f4592914c61e4e946cad9d0eba75c

myrinzic, 2023-01-17:
Leantime ticket #112

sparklestheunicorn, 2023-10-04:
This should be all set

closed 2023-10-04 by sparklestheunicorn