FourThievesVinegar/solderless-microlab #25: Finish hardware readme

mis4mike, 2021-02-07:
There is currently a partial README.md file in the hardware directory of the project. This needs to be updated with:
- A complete list of parts (specs, manufacturer numbers, where to get them etc.). This includes types of syringes, tubes, printing recommendations, and all electronics. Preferably with pics.
- A complete list of assembly steps with pics.
- If anyone has a completed prototype and can provide any specs, that would be awesome!

myrinzic, 2023-01-17:
Leantime ticket #111

myrinzic, 2023-01-17:
Leantime Ticket 111

sparklestheunicorn, 2023-10-04:
This is largely complete (more details and pics are required in docs/operation.md). Separate issues can be made for updating particular parts of the documentation.

closed 2023-10-04 by sparklestheunicorn