FourThievesVinegar/solderless-microlab #31: 500 Error from backend when accessing /select/option/Done

mis4mike, 2021-02-08:
I get this error when I proceed on a development machine through the aspirin recipe. It seems to happen after the final manual step.
```
GET http://localhost:5000/select/option/Done 500 (INTERNAL SERVER ERROR)
```
The error on the python client is:
```
[2021-02-07 19:29:33,159] ERROR in app: Exception on /select/option/Done [GET]
Traceback (most recent call last):
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/connection.py", line 552, in connect
    sock = self._connect()
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/connection.py", line 609, in _connect
    raise err
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/connection.py", line 597, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/celery/backends/redis.py", line 127, in reconnect_on_error
    yield
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/celery/backends/redis.py", line 177, in _consume_from
    self._pubsub.subscribe(key)
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/client.py", line 3528, in subscribe
    ret_val = self.execute_command('SUBSCRIBE', *iterkeys(new_channels))
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/client.py", line 3416, in execute_command
    self.shard_hint
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/connection.py", line 1185, in get_connection
    connection.connect()
  File "/home/mike/workspace/solderless-microlab/backend/env/lib/python3.6/site-packages/redis/connection.py", line 557, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

During handling of the above exception, another exception occurred:
```

thenixlabs, 2021-02-22:
Looks like you don't have the redis server running. On the pi you would need install the following packages:

sudo apt -y install screen git redis-server python-celery-common python3-flask python3-pip  python3-rpi.gpio python3-serial

"redis-server" is what you are missing. If you are running on a platform not based on debian take a look at https://redis.io/

mis4mike, 2021-02-28:
Thanks! Adding this to the setup documentation.

closed 2021-02-28 by mis4mike