FourThievesVinegar/solderless-microlab #74: Manifold needs a hole for the temperature probe.

mis4mike, 2022-07-18:
Currently there's no dedicated hole in the manifold for the temperature probe. Its already tight jamming the temp control tube into the manifold top, so we could use a dedicated place for it.

myrinzic, 2023-01-17:
Leantime Ticket 92

sparklestheunicorn, 2023-11-13:
> > Currently there's no dedicated hole in the manifold for the temperature probe. Its already tight jamming the temp control tube into the manifold top, so we could use a dedicated place for it.
> 
> create a special repository and store it there

We do have a parts repo: https://github.com/FourThievesVinegar/Parts/tree/master/v5 - What we need is an updated CAD file for the parts that includes this new hole.

sparklestheunicorn, 2024-04-25:
The v0.6 manifold core has many more holes than the previous reactor design. This is complete.

closed 2024-04-25 by sparklestheunicorn