FourThievesVinegar/solderless-microlab #80: SPIKE: How do we get recipes onto the microlab?

sparklestheunicorn, 2022-08-28:
Currently, the process of adding a recipe to the Microlab is to copy a file onto the SD card or download it manually from the internet. It would be nice if there were a more integrated way to add a recipe, rather than "Get a file and put it in the recipes directory"

0x42red, 2022-08-28:
I imagine we could do the following.

Web GUI intakes recipe -> API validates it
The validation can check for the following
  - Does the recipe have the valid data in it
  - Does the recipe run
  - Does the recipe have an exit condition
  - Does the recipe break any safety bounds(temp probes are rated for N degrees and they want to use N+20)
  - additional thinking on this could be good

Once the code is created the GUI could send another API request to take that same recipe and return a flowchart for the user of the recipe. This will allow them to confirm the steps look how they expect.

Both of these steps can give a better UX but also prevent broken recipes from getting to users/being ran with runtime errors.  Additionally since recipes are live Python code instead of just JSON it might make sense to allow for simulation tests.  For example they can provide a list of steps and the selected decisions with the expected outcome.  This can allow recipes with branches to be quickly simulated instead of attempting to calculate all the permutations of a recipe.

myrinzic, 2023-01-17:
Leantime Ticket 31

closed 2024-04-25 by sparklestheunicorn