FourThievesVinegar/solderless-microlab #83: Migrate to 3MF from STL for hardware files

0x42red, 2022-08-28:
Currently we use STL files for the 3d files.  We should export a 3MF version of the file for use instead.  With 3MF units of measurement are baked into the format so reproductions should be more accurate as well as some other features that might matter to the project in the future.

https://www.3dnatives.com/en/stl-vs-3mf-280620224/

0x42red, 2022-08-28:
@sparklestheunicorn in the log it looks like you made the last change to the STL.  Do we have access to the CAD project files still?  Would that make sense to also put them in the repo?

myrinzic, 2023-01-17:
Leantime Ticket 91

sparklestheunicorn, 2023-10-04:
Alas we do not have access to the original CAD files - they will need to be re-cadded.

sparklestheunicorn, 2024-03-06:
Parts have been re-cadded and [we now have](https://github.com/FourThievesVinegar/microlab-parts/tree/master/v5) both STL and STEP files

closed 2024-03-06 by sparklestheunicorn