FourThievesVinegar/solderless-microlab #178: As a user and Developer, I want the HDMI and small display to both work at the same time

sparklestheunicorn, 2024-04-27:
Currently when you install the LCD drivers they disable the HDMI port. It's not clear how to un-do that. Technical research on the feasibility and approaches would be really great.

phaseloop, 2024-09-08:
I did similar research some time ago. TLDR: It's ugly, hacky, difficult, board dependent and may bring more problems than needed:
The only solution I found:
https://forums.raspberrypi.com/viewtopic.php?t=306624

Raspberry PI becomes a liability due to their politics and pricing. There are now more powerful and cheaper alternatives.

As far as I see you use 4-5 GPIO ports to control relays. This can be moved to CNC shield as CNC controllers (grblHAL included) support M64-M65 commands which are used to control additional GPIO pins on the board (exactly for scenarios like that).

So technically -  a next version could just wire all IO and motors to CNC shield (whatever board is supported by grblHAL) and just expose USB ports/hub. Then you use whatever PC/laptop/Intel N100/raspberry you have.

This essentially halves the cost of microlab controller because you can repurpose any computer laying around and gathering dust instead of buying raspberry (which I remind is prone to semiconductor shortage and delivery problems as it happened during covid).

sparklestheunicorn, 2025-10-20:
DSI port is how we recommend people do it now.