FourThievesVinegar/solderless-microlab #185: PARTS: As a user I want different or expanded BOMs for different regions of the world

sparklestheunicorn, 2024-05-08:
Currently the Microlab BOM is written by and for audiences in the US. Replicators in the EU have reported difficulty obtaining certain parts and needing to use replacements. It would be nice to capture a variety of alternatives folks have used to make it easier for everyone.

sparklestheunicorn, 2024-09-09:
Potential [duplicate of this issue](https://github.com/FourThievesVinegar/solderless-microlab/issues/244). Leaving them both for now, but closing one probably means we can close the other.

phaseloop, 2024-09-10:
I did a review of BOM - except mason jars nothing seems extraordinary for EU market (most of stuff in BOM sourced from Aliexpress anyway).

1/4'' copper pipe can be replaced with 6 mm copper pipe metric standard.  Barbed tee fitting from McMasterCarr is available in most construction shops (think HomeDepot) on garden section or in motorcycle shops (for fuel lines probably).

If you can check in user reports what parts were troublesome, I could give some guidance.