FourThievesVinegar/solderless-microlab #193: Research alternatives to copper coil heat exchanger

sparklestheunicorn, 2024-05-31:


phaseloop, 2024-09-08:
Silicone hose wound around 3d printed frame and secured with zip-ties? Popular in helix RF antennas:

https://www.thingiverse.com/thing:4980180

Thermal conductivity of silicone is like 800 times worse than copper but it's cheap and easy to make so maybe in this application it can be a "good-enough" trade-off.

While I suppose biggest pain-point of copper coil is actually winding it - maybe 3d printed frame + thin copper brake line can be a good and easy way to do it.

Unless there is a different need to get rid of this heat-exchanger.

phaseloop, 2024-10-28:
Removing need for a heat exchanger: https://github.com/FourThievesVinegar/solderless-microlab/issues/276

sparklestheunicorn, 2024-11-07:
We're already concerned about efficiency losses from the copper, so moving to silicone probably isn't the move. There are also automotive aluminium heat sinks (~1/2 the heat conduction of copper) that are pretty cheap and available. Honestly, bending the coil around a cylindrical jig isn't that hard, but it would be nice to have other options. 

Directly heating the jacketed region as suggested in #276 would be great if we can get it working :+1: