FourThievesVinegar/solderless-microlab #215: SPIKE: PARTS: As an assembler, I want to know whether DIY peristaltics are good and which ones are best.

sparklestheunicorn, 2024-07-08:
At ~$30/ea in 2023, the three peristaltic pumps are a major part of the cost of the MicroLab. In addition to being costly, they are also relatively limited in terms of the number of vendors selling them to the public. It would be good to look into alternatives and their reliability.

There are a number of 3d-printed peristaltic pump designs out there ([one example](https://www.myminifactory.com/object/3d-print-peristaltic-pump-63137)). An exploration of the various options and tests of their reliability would be helpful in deciding whether we think we want to encourage our users to go with printed pumps, and if so, which ones?

phaseloop, 2024-10-10:
Here is the curated list of available pump designs from thingiverse and printables.

**Also it's worth considering a different approach - pressing a syringe cylinder to extract given volume of liquid instead of pulling it using pump.** The downside of it is the limit of syringe volume while with pump we can connect anything to the input side.

Used assumptions:

- no print-in-place designs, they need good printer tuning and tolerance. Also when they fail you probably need to print a new one from scratch.
- only pumps using steel bearings as pressure elements to have as biggest MTBF as possible and build consistency (i.e pump calibration being a bitch)
- pump form allowing it to mount/adapt it to microlab. No standalone big pumps or weird designs
- stepper motor pumps only, no weird steppers - only Nema 17.
- as simple to print and build as possible - no planetary gears, etc.

**Pump designs**

**Editor's choice!**

- https://www.thingiverse.com/thing:4692199

Heavy-duty, well designed pump with adjustable hose tension. Includes source scad files for further modifications (no other design on the list probably does that).

Uses 6 mm tubing.

**Other**

- https://www.thingiverse.com/thing:4255438

Easy and nice design - uses 3 mm tubing. Pump intake seems to be integrated into housing (??) which seems to be weird.

- https://www.myminifactory.com/object/3d-print-peristaltic-pump-63137 

(OG design linked in this issue)

- https://www.printables.com/model/248041-growflow-nema-17-peristaltic-pump

Similar design but uses less popular needle bearing

- https://www.thingiverse.com/thing:1134817

Nice and secure design but needs some minor fixes to tube guide to prevent it from moving and keep the precision.

- https://www.printables.com/model/818373-nema-17-peristaltic-pump-water-pump-measuring-pump

ok-ish but tube seems to not be secured properly and there is a risk it will fall out during operations

- https://www.printables.com/model/129756-peristaltic-pump-nominal-diameter-o8mm-for-fluids-

ok-ish but the bearing mount gives me anxiety and I doubt it will work more than 10 minutes. Someone tried to adapt aluminium pump to 3d printing.

- https://www.thingiverse.com/thing:3346506

Easy and clean design but risk of tube falling out

- https://www.thingiverse.com/thing:3109449

Nice and clean design but bearings are not secured with screws and keeping pressure between pump rotor and motor shaft is the single point of failure.

sparklestheunicorn, 2024-10-13:
Thanks @phaseloop! We're running this by some of the CAD design folks.

phaseloop, 2024-10-13:
I have a pretty long and solid experience in FreeCAD - I would love to join discussions if that's possible.

sparklestheunicorn, 2024-10-21:
@phaseloop please drop us an email at volunteerⒶfourthievesvinegar.org and we'll put you in touch.

with-eight-hands, 2024-12-19:
Thank you phaseloop for compiling this list!

I have tested https://www.thingiverse.com/thing:4255438 and found it to be simple and relatively functional. Comments:
- The barb connector inlet/outlet is not required; the design can be used with one continuous piece of tubing.
- The bearings aren't quite wide enough for the specified tubing. (The tubing becomes wider when squished flat.) It is more reliable if the middle part of the rotor is flipped so that the bearings are in pairs directly adjacent, used for a single tube.
- Inserting the tubing and rotor relies on the tubing being quite stretchy. To accommodate stiffer tubing (e.g. PTFE), it would be better to have slots rather than holes on the side of the housing.

with-eight-hands, 2024-12-22:
I made a pump design inspired by the design linked above. It uses pairs of bearings to accommodate tubing with flattened widths up to 8 mm, and it has slots so that tubing stretching is not necessary. The amount of tube compression is changed by varying the distance from the center to the bearing mount holes. (This isn't very clever, but it keeps the design simple.) I have tested it with PTFE tubing (11G, 2.41 mm ID, 3.01 mm OD) and silicone tubing (3 mm ID, 4 mm OD) and it seems to work OK.

[image]

STL files here include parts for rotors with dimensions that yield 0.55, 0.70, or 0.85 mm clearance between the bearings and housing walls. 0.55 mm clearance worked reasonably for PTFE tubing with a nominal 2\*wall_thickness of 0.60 mm. 0.85 mm clearance worked for silicone tubing with a nominal 2\*wall_thickness of 1.0 mm.

[pump-prototype.zip](https://github.com/user-attachments/files/18222714/pump-prototype.zip)

phaseloop, 2024-12-23:
Cool! Have you seen the design uploaded recently? It also uses pair of bearings.

https://github.com/FourThievesVinegar/microlab-parts/tree/master/peristaltic-pump

We could really use a feedback on that design, especially about what you are writing about teflon tubes - maybe we can use what's best from both of them.

with-eight-hands, 2024-12-23:
Oh, I did not. Very nice! Unfortunately I won't be able to do more prototyping any time soon, but I can give some impressions.
- Overall I expect that should work fine with teflon tubing, given the right spacing. Changing the bearing assembly dimensions like I did should work fine.
- The swing-away design looks like it could be easier to load than the design I used, which requires some awkward sliding.
- Having the top bracket fixed onto the design has a downside: you must fish the end of the tube through the pump in order to load and unload it. If the tubing has been used with chemicals, you will need to clean the tubing or carefully cover the ends before unloading in order to avoid contamination. Poor chemical hygiene is bad for tools as well as bad for people (even a tiny drop of hydrochloric acid from mostly empty tubing can cause horrible rusting). I suppose for this design that just means unscrewing the housing / top bracket during every load and unload, which isn't too big a problem.

Overall it looks like slightly more work to assemble and to load the tubing (assuming we are not passing the tubing end through the pump), but I'd guess it should work OK, even with teflon.