FourThievesVinegar/solderless-microlab #242: [BUG] Permission Denied to /etc/microlab/microlab.ini

zenny, 2024-09-05:
Hi,

Thanks for posting an interesting repo.

Trying to build a development server to try with `python3 main.py`.

```
$ python main.py
Traceback (most recent call last):
  File "/home/user/Downloads/solderless-microlab/backend/main.py", line 11, in <module>
    import config
  File "/home/user/Downloads/solderless-microlab/backend/config.py", line 125, in <module>
    microlabConfig = MicrolabConfig()
                     ^^^^^^^^^^^^^^^^
  File "/home/user/Downloads/solderless-microlab/backend/config.py", line 24, in __init__
    makedirs(path.dirname(configFileName), exist_ok=True)
  File "<frozen os>", line 225, in makedirs
PermissionError: [Errno 13] Permission denied: '/etc/microlab'
```

Since the entire build was built as a user permission, why running the `main.py` searches for an inaccesible config in `/etc/microlab/microlab.ini` making it unsafe? I could not find `microlab.ini` examples in this repo either.

Cheers,
/zenny

sparklestheunicorn, 2024-09-06:
In current production environments (on a RPi), running the python script using sudo is necessary to access the GPIO pins, so that's the way the software and scripts have been designed. We have made some efforts to make everything run as a regular user, but it hasn't been high enough priority compared to other recent features and release activity.

I'll leave this open until we (a) figure out how to remove the need to run the backend scripts as super user or (b) determine that's the only way to do things and see if there's a workaround for development on non-pi machines.

zenny, 2024-09-06:
> In current production environments (on a RPi), running the python script using sudo is necessary to access the GPIO pins, so that's the way the software and scripts have been designed. We have made [...]

@sparklestheunicorn Thanks for the reply.

What I assigned the ownership of the following directories to the user as an alternative way than using `sudo` itself:

```
sudo mkdir -p /etc/microlab
sudo chown -R $USER:$USER /etc/microlab
sudo chown -R $USER:$USER path/to/this/repo/backend/env
sudo chown -R $USER:$USER /var/log/microlab
sudo chown -R $USER:$USER /var/lib/microlab
```

to make $USER to run the script, but not ideal, imho, securitywise. 

Thanks for the wonderful work and look forward to changes you are planning to overcome the bottleneck. 

Cheers,
/zenny

sparklestheunicorn, 2025-09-09:
This should be resolved with the new imager repo

closed 2025-09-09 by sparklestheunicorn