FourThievesVinegar/solderless-microlab #244: PARTS: As non-US user I want to use non-mason jar glassware

phaseloop, 2024-09-08:
Outside US, mason jars are either not available, available in limited size choice or way overpriced. This makes building microlab much more difficult or impossible (depending on region/country).

Solution: Create a 3D adapter to use regular ("European style" [?]) jars - those are used to package food and are available worldwide even as recycled glass. Standardized lid diameters are 33, 43, 53, 59, 66, 77, 82, 100, 110 mm. Outside jar should use 82 mm which is most popular (need to decide internal one diameter).

**Note:** Non-mason canning jars used in US (judging from the photos) use different thread sizes and lids than the rest of the world

The issues with regular jars are:

- there is no "neck" below lid as in mason jars so there is no way to attach them to the device using flange.
- lids are made of carbon steel and are heavily prone to rusting once internal plastic coat is damaged - so cutting a center hole for reactor manifold lid is not an option (aside of that - it's pain in the ass to cut).

Solution: 3D printed jar lid (adapted to thread which I believe is ISO standardized for jars) - with center hole matching reactor manifold and external features allowing it to work with flange and hand the jar on the device. If not - separate flange should be designed to match lid features that in tandem allow this to secure jar in the device.

This issue will evolve - I'll try to design those lids in CAD but I have ADHD and focusing problems so I'll be really thankful for someone to respond here once in a while and remind me of it. Any feedback is welcome - I love what you do!

phaseloop, 2024-09-08:
Edit: I'm looking at the photos of the device and I see that jar is standing on a table next to the microlab. Can someone here say what is the role of a flange here?

sparklestheunicorn, 2024-09-09:
This looks like a potential [duplicate of this issue](https://github.com/FourThievesVinegar/solderless-microlab/issues/185) (but with more detail :+1: ). Leaving both open for now and linking them.

We had a replicator in the EU produce another manifold for jars available there. Since then, we have redesigned the manifold however, so it might be good to choose appropriate jars for the European market, and [localize this directory](https://github.com/FourThievesVinegar/microlab-parts/tree/master/v6).

This would mean (to start) creating a new `europe` subdirectory in the `v6` directory, and creating any adapted part files there. We would also want a list of localized off-the-shelf parts added to a new tab in the parts list spreadsheet for each market. 

Finally, we would need to add an addendum document for each new region and mention them in the appropriate places (when discussing sourcing the parts and prominently in the documentation for any module that requires modifications.

phaseloop, 2024-09-18:
An update - co called "2030 type thread" most popular in jars around the world is awful to model for 3D printing because the short and thin plastic thread in hot environment will not be reliable at all. Maybe there is a way to attach the lid different way - but in the meantime I've designed manifold lid to be used in "Weck" system jars - secured by metal clamps used by default in this system.

While Weck jars and not exactly the cheapest ones - they are much more popular across the world. AFAIK reactor core can be adapted too to Weck system when a matching set of jars is found.

System picture for reference. After some testing I'll post the PR with the lid.

[image]

with-eight-hands, 2024-12-19:
See #292 for an example using a media bottle with DIN GL45 threads as the inner vessel. 3D-printed threaded lids seem to work fine with this thread profile. Wider threaded mouths are defined in the same standard, but they are less commonly available, so the outer vessel may require a different solution.