FourThievesVinegar/solderless-microlab #245: Microlab should be independent from Raspberry PI

phaseloop, 2024-09-08:
I extracted part of my comment from #178 because I think it deserved another issue:

Raspberry PI becomes a liability due to their politics and pricing and recent versions become more expensive than a regular mini-pc. There are now more powerful and cheaper alternatives.

As far as I see microlab uses 4-5 GPIO ports to control relays. This can be moved to CNC shield as CNC controllers (grblHAL included) support M64-M65 commands which are used to control additional GPIO pins on the board (exactly for scenarios like that).

So technically -  a next version could just wire all IO and motors to CNC shield (whatever board is supported by grblHAL) and just expose USB ports/hub. Then you use whatever PC/laptop/Intel N100/raspberry you have.

This essentially halves the cost of microlab controller because you can repurpose any computer laying around and gathering dust instead of buying raspberry (which I remind is prone to semiconductor shortage and delivery problems as it happened during covid).

0phelia36, 2024-09-10:
I'm going to take a look into adding software support for controlling gpio pins via grbl. The microlab can already be independent from the raspberry pi specifically if you install the microlab software yourself on another SBC, and create a config file for the SBC you're using. We've created a file for the the librecomputer potato that you can see [here](https://github.com/FourThievesVinegar/solderless-microlab/blob/master/backend/data/hardware/controllerhardware/AML-S905X-CC-V1.0A.yaml) that could be used, and can be referenced if anyone wants to try this with another raspberry pi clone.

phaseloop, 2024-09-10:
Thank you, I didn't know that.

As for grbl - while only newer grblHAL (for 32 bit boards) support "native" pin control, with older arduino + CNC shields it would be possible to use pins and commands used for coolant pumps and laser air-assist (M7/M8/M9 commands for example) - then just wire microlab relays to those pins.

I have a background in electronics and CNC so let me know if you need any help :).

0phelia36, 2024-09-12:
I think that only gives 3 pins though? At least with the cnc shield we're using. Unless I'm missing something there's: spindle enable + spindle direction + coolant enable. I tried it out on mine and M3 and M5 didn't seem to change the state of spindle enable either, so really only 2, which would not be enough to be the only source of gpio ports for running the microlab.

michal-rogala, 2024-09-13:
Can you try "M3 S10000"? - grbl usually does not enable spindle power without complete speed settings.

M7 command for coolant pump mist is by default disabled in BRBL - it should drive pin A3 (or A4/A5) even if it's not exposed on CNC board - it's available on the underlying board.

Should be not a big problem to make a compiled version of grbl (with enabled M7) and share on github to just flash the board and not bother users with firmware compilation anyway.

phaseloop, 2024-09-13:
Can you try "M3 S10000"? - grbl usually does not enable spindle power without complete speed settings.

M7 command for coolant pump mist is by default disabled in grbl - it should drive pin A3 (or A4/A5) even if it's not exposed on CNC board - it's available on the underlying board.

Should be not a big problem to make a compiled version of grbl (with enabled M7) and share on github to just flash the board and not bother users with firmware compilation anyway.

0phelia36, 2024-09-15:
"M3 S10000" also doesn't work, the spindle enable pin just stays high. Tried it also with "M3 S0", "M4 S0", and "M4 S10000", with every one of those followed by M5 and nothing changed the state of the pin. M3 and M4 are changing the state of spindle direction. Not sure what the issue could be with spindle enable.

phaseloop, 2024-09-16:
It may be in "laser mode" when the laser only gets enabled when x/y axis are moving to avoid fire risk of laser stopping and burning a hole through the table.

Can you dump the config using `$$` ? Also you can try if `$32=0` command changes the behaviour.

0phelia36, 2024-09-17:
$32=0 also didn't change anything.
Config:

> 
> $$
> $0=10
> $1=25
> $2=0
> $3=0
> $4=0
> $5=0
> $6=0
> $10=1
> $11=-0.000
> $12=0.002
> $13=0
> $20=0
> $21=0
> $22=0
> $23=0
> $24=25.000
> $25=500.000
> $26=250
> $27=1.000
> $30=1000
> $31=0
> $32=0
> $100=250.000
> $101=250.000
> $102=250.000
> $110=240.000
> $111=240.000
> $112=240.000
> $120=10.000
> $121=20.000
> $122=10.000
> $130=200.000
> $131=200.000
> $132=200.000
> ok

phaseloop, 2024-09-17:
Ok, I've reviewed the grbl source code and the mystery is solved why spindle enable pin is always 5V.

There is a mismatch between CNC shield design and newer grbl versions for the sake of spindle speed control. Real spindle enable pin is now the "Z+" endstop on the board. This can be easily changed back in config.h.

As grbl is not developed anymore (replaced by grblHAL for 32 bit boards) and it only supports 328p microprocessor anyway, we can safely fork their repository - change default configuration to match desired behaviour, release precompiled .hex file for people to download instead of making them compile it.

Should not be that hard later to gut some unnecessary code (as we don't do CNC anyway) to make some space in flash and implement M65 code to toggle more pins - like Y+, X+, Hold, Resume, etc.

This would be forward compatible if the future microlab will switch to grblHAL CNC boards when they became cheaper and more popular (which already happens) - as those boards already implement M65 and pin maps as outputs to control.

0phelia36, 2024-09-20:
Is that something you're capable of and interested in doing? That's outside of my skillset and capacity to work on at the moment, but I think would be good to have. Having all the gpio control done through the arduino would make it much easier to support other SBC's with the current build since we wouldn't need to go through creating config files for each one.

phaseloop, 2024-09-25:
Yeah, I can do it. We need to solve one issue in the meantime - how to easily connect relay pins to the CNC board holes like spindle enable, Y+, etc. As soldering is a no-no - I guess there needs to be another way.

0phelia36, 2024-09-26:
Cool, I've forked grbl [here](https://github.com/FourThievesVinegar/grbl) and have given you access to the repo. And unless I'm missing something I don't think the connections between the arduino and the relays would have to be any different than current connections between the pi and the relays?

phaseloop, 2024-09-26:
Thanks!

Yeah, you're right - I used to work with some really old CNC shield design, I see newer versions have proper pin headers in a row, so that solves a lot of issues.

0phelia36, 2024-10-15:
Yeah I can test it, though I won't be able to until probably this weekend.

0phelia36, 2024-10-21:
I tested it and everything seems to work fine, so I'm going to work on adding support in the microlab for it

0phelia36, 2024-11-28:
@phaseloop PR #282 adds control of gpio pins through grbl. You can use the arduino-only-no-SBC controller hardware file in that PR to test only using the arduino, though our version of grbl will need to be modified to add a 4th pin, it currently errors trying to setup the nonexistent fourth pin. If you change the "BCM_26: 4" line in that file to to "BCM_26: 3" then you can test that the pin control is working, though the 3rd pin will be behave strangely since two different things are then controlling it.

sparklestheunicorn, 2024-12-17:
This may not work for our purposes but adding it here in case anyone wants to investigate the option of using an old Android smartphone instead of a SBC: https://hackaday.com/2024/12/16/using-a-smartphone-as-a-touchscreen-for-arduino/

phaseloop, 2024-12-23:
@0phelia36 - Thanks, I implemented pins 4 and 5 in GRBL (spindle enable and direction). I need to test it and I'll create a new grbl release.

phaseloop, 2025-01-04:
@0phelia36 - I just released grbl v0.0.2 -we now support pin 4 and pin 5 too