FourThievesVinegar/solderless-microlab #249: As a developer, user, and future-proof person, I want a desktop UI

sparklestheunicorn, 2024-09-12:
A little bit of responsive design could go a long way. Both development and [future expansion to run on a standard laptop](https://github.com/FourThievesVinegar/solderless-microlab/issues/245) would be a lot easier.

Some requirements include:
- Keep all existing functionality and compatibility with the small touch-screen
- Don't substantially impact performance on the Pi

Some additional elements to add on a larger screen:
- Logs
- Expanded status
- Next step and/or recipe overview

mushkevych, 2025-04-28:
Below are workaround to support larger 800x480 display:

1. Update the `app.css`:
```shell
vim.tiny /home/thief/solderless-microlab/gui/src/styles/app.css
```

and update following blocks:
```css
.lcd-wrapper {
  position: relative;
  width: 800px;
  height: 480px;
  margin: 0 auto;
  background: white;
  overflow: hidden;
}

.page {
  padding: 0.5rem 1rem;
  height: 432px;
  overflow-y: auto;
}
```

2. Rebuild UI with enabled Internet access. 
Note - when i was rebuilding without Internet access the UI Layout broke. 
```shell
cd /home/thief/solderless-microlab/gui/
yarn build
```

3. Restart the service
```shell
systemctl --user restart microlab-serve.service microlab-browser.service
```

Bedrovelsen, 2025-05-09:
how about a ui that does not need to be built made of semantic html and vanilla classless css for easy responsive goodness? Also can be very small I would guess in processing and file size as just text. Maybe no js even needed depending

mushkevych, 2025-09-08:
@sparklestheunicorn can this ticket be resolved after #344 was merged into the `main` ?