FourThievesVinegar/solderless-microlab #276: PARTS: Use small ceramic heater and limit fire risk of running heat exchanger

phaseloop, 2024-10-28:
Most common failure mode of a relay is being stuck in "closed" position. 

Having 500W water heater in a small jar is a big fire risk - in case something goes wrong (relay or software issue) heater will evaporate water fast and then start to melt itself and surrounding parts, starting fire.

An alternative is to use small ceramic heater used in 3D printers (cost is $3-6). Heater powers range from 20 to 100W and after simple waterproofing it can be submerged in the outer reactor jar. In case something goes wrong - heater has contact only with non-flammable wire insulation and glass and will burn through heating material and stop working before doing more damage.

There are also significant benefits to this solution:

- reduced fire risk (as mentioned above)
- simplified build time and BOM as hot side heat exchanger is not needed anymore
- not relying on mains voltage and being able to run microlab off battery or small power inverters
- possible better efficiency than heat exchanger

Keeping hot side circulation pump is a matter of some tests and discussion - whether water mixing to distribute heat around jar is needed or regular heat convection will be enough.

sparklestheunicorn, 2024-11-07:
An example [product link](https://www.amazon.com/dp/B07YFBRKZ4). Seems like waterproofing and general heating effectiveness are the main questions here.

phaseloop, 2024-11-08:
Effectiveness can be improved easily by making a ring out of copper pipe (from heat exchanger) and putting the heater inside the pipe on one end. I suppose 6 mm heater diameter is just right for it to fit.

dallinbeutler, 2025-02-07:
apparently brewers use  this [immersion heater ](https://www.amazon.com/DERNORD-Immersion-Cartridge-Heater-Tublar/dp/B0BC752LPS?pd_rd_w=pHuoS&content-id=amzn1.sym.cd152278-debd-42b9-91b9-6f271389fda7&pf_rd_p=cd152278-debd-42b9-91b9-6f271389fda7&pf_rd_r=EK8MB96SZMRG5S327DK7&pd_rd_wg=BidcH&pd_rd_r=786f6788-8a1f-41ea-aef4-292a8957afac&pd_rd_i=B0BC752LPS) made of stainless steel that could be part of the reactor core stirring lid insert and be an easy plug n play 

cheaper, but more work would be a [ptc heater](https://www.amazon.com/Heating-Thermostat-Element-Temperature-Resistant/dp/B07VWP5LLJ?crid=3CYKN0R0SLA9T&dib=eyJ2IjoiMSJ9.hWinpmcqSU--3l81wofUxjkJqYWFLJil2yiQmMpg4ebTRqirZuHhcn8IzUG-nrPTyTQh5NDd0fJsL1mboLozBpqD07m7nHYrK2XyukMlpZZHA3j-a0P7u5-wtDelv_nN3L0SdHVub54Xou0au5DEXjTKyJ9eaEcktmiE0WEXkqstjUNB6SK7pMPlgDot7KLUwLZzNjniM82I5JkkZ6baeIOfnayyb21RuhXXZ48MGtGBJ0ouMLMVPiuza1J4qDdxfYMu4lRagK_LQzvvzldM0arfjOrRsRhOWKnt6xPV5smFzDIWOzkwJGu6iwcVihsYzt4moLDv6YlF7MjGd2BP1BBHv36JOD8IZKGPXSxRtvJp2MGonvB0N3gTm4CMUELeyc41oVtgb6whXOCJuL7qjljI3rjBXibe2R_d4Bn5Z_90CNFCsnerrVl3cFFPr3K7.hdSdtWQsyqbv49UPU6Sune4K--Wj6KuLQdpuvD0bv6A&dib_tag=se&keywords=ptc+heating+element+12v+waterproof&qid=1738910101&sprefix=ptc+heating+element+12v+waterproof%2Caps%2C113&sr=8-24#customerReviews)  with the added benefit of automatically shutting off at specific max temp depending on the component.
as for waterproofing, could protect it with a high enough temp plastic sleeve of a sort, and inserting it directly in the outer circulation bath