FourThievesVinegar/solderless-microlab #287: Can't run the backend with docker

suspiciousRaccoon, 2024-12-03:
heya!

I tried running the project with docker but there are a few problems.

1.  `TypeError: 'type' object is not subscriptable`
This is the first error I encountered. A quick google [search](https://stackoverflow.com/a/63460173/21430041) led me to this problem originating from the type hints. Older python versions had a different API, so they're not compatible with the dockerfile's python version. Upgrading the python version to 3.10 fixed this.
2. Missing dependencies
`backend/api/`

0phelia36, 2024-12-04:
Hello! Thanks for the issue. 

The gpiod error you're running into is due to the fact that gpiod is not supposed to be installed via pip at the moment and instead has to be installed as a system package via apt to be used using `sudo apt install python3-libgpiod`. If you're just trying to run the code for development or testing you can instead go into the frontend, go to settings, and change the value of the lab hardware config dropdown to "simulation-pi" and that error will be avoided as it won't attempt to import and use gpiod to control the state of the gpio pins, and will instead use stub functions for gpio control that basically do nothing.

I don't think anyone has run the project with docker in a while so it might take some work to get that fully running, at the very least the docker-compose file will need to add volumes mounted for /etc/microlab and /var/lib/microlab for the microlabs configuration and recipe data to persist, not sure if there's anything else that's changed that might cause issues.

phaseloop, 2024-12-04:
I'll try to run it and fix those errors. @suspiciousRaccoon this "No such file or directory" error may be caused by GPIO device being unavailable. Are you running this on raspberry? If so - did you enable raspberry GPIO support/driver and uncomment docker-compose path mapping?

phaseloop, 2024-12-27:
Will be fixed by https://github.com/FourThievesVinegar/solderless-microlab/pull/296

phaseloop, 2025-01-06:
@suspiciousRaccoon  - please test the fix (merged to main) and close the issue or report other bugs if you find them. Thanks :)

suspiciousRaccoon, 2025-01-10:
Hi!

Sorry for taking a while. I found a tiny issue but it should be fixed by #303. That aside, docker seems to work! c:

closed 2025-01-10 by suspiciousRaccoon