FourThievesVinegar/solderless-microlab #292: Use materials with better chemical compatibility (teflon and borosilicate) for the inner reactor vessel (prototype included)

with-eight-hands, 2024-12-19:
[Please let me know if there's a better place to put things like this!]

The current design has the reactor contents in contact with a mason jar (soda lime glass) and the 3D-printed reactor core (probably PETG or PLA). This is undesirable because the reaction mixture may degrade these materials, causing mechanical failure (bad) or contamination of the reaction mixture (even worse). From a safety perspective, I believe this should be a priority, and based on some preliminary work, I think this is achievable in the near term.

Avoiding soda lime glass is straightforward; simply use the lab equivalent of a mason jar. Borosilicate media bottles are widely used for chemical storage and rated for autoclaving. They are available with a DIN-standard screw profile (e.g. GL45), so designing new lids isn't too hard.

Avoiding 3D-printed parts in contact with the reaction mixture involves slightly more effort. PTFE (Teflon) is one of the few polymers that is compatible with most common lab chemicals. It is easy to machine, but conventional machining might be considered inaccessible for many users. Luckily, similarly to how the classic mason jar lid design is separated into two parts, a reactor lid can be designed around a simpler PTFE disc and a more complex 3D-printed screw closure that connects to the external vessel and the motor. The minimal PTFE design would require cutting the circular outline and drilling 4 holes (1 for the stir rod and 3 for the reactants). Because the sealing* surface is the bottom face of the disc rather than the edges, the circular outline does not need to be precise and can be cut with a hacksaw. The holes for the reactant tubing do not need to be precisely located or angled and can be drilled by hand. The stir rod hole is the most critical. It should be perpendicular and centered, so would ideally be done with a drill press, but there is likely enough wiggle room for hand drilling.

I made a prototype based on the above ideas. The bottle (~10 USD), PTFE (~10 USD), and filament (~5 USD) represent pretty modest costs. Media bottles are also more commonly available internationally, partly resolving #244 . As you can see the PTFE disc outline is quite rough, since I cut it by hand, and yet that doesn't impact performance in any way. The glass/PTFE contact happens on the lower surface of the disc, not the edges.

[image]

[image]

The single 3D-printed piece clamps the PTFE disc to the bottle.

[image]

It also provides the motor attachment and flange to clamp to the outer mason jar.

[image]

[image]

Remaining issues:
- The choice of GL45 necks limits the room to work. Addition of glassware adapters or more than 3 reactants could be tricky. I don't see this as a major concern. GL80 media bottles (25 mm wider) exist, but they are harder to find especially for bottles <500 mL.
- The operating temperatures and safe ramp rates are still limited by the 3D print material and the outer mason jar. Replacing the outer mason jar may be feasible (especially if it doesn't need to be transparent), but making the screw-thread assembly without 3D printing or conventional machining might not be realistic.

Hopefully this gets other people thinking about how we can make this apparatus safe to use. I personally think this approach is accessible.

*Please don't try to do chemistry in a fully sealed glass vessel! Unintentional bombs are no fun.

with-eight-hands, 2024-12-19:
Here's the STL of prototype printed adapter. I might upload the FreeCAD source after some tweaks.

[GL45-to-mason-with-motor.zip](https://github.com/user-attachments/files/18203594/GL45-to-mason-with-motor.zip)

phaseloop, 2024-12-23:
Hey! I've been gathering concepts for the redesign of reactor manifold and your idea is really cool.

I suppose GL45 is the way to go, but we need to keep reagent tubes diameter pretty large - otherwise part of reagent volume will stay there (as in capillary tubes).

As for PTFE disk - do you think 3d printing a ring with special shape and wrapping it with PTFE tape would work too? There would not be holes for hoses but rather pockets around center hole so it's easy to wrap a tape around it.

As for outer jar - so far it seems we need to stick with "regional" variants - like mason jar, Weck system, etc. While it's impossible to 3d print a lid for an "european" jar, I'm experimenting with drilling center hole in a metal lid using step drill and secure the rest of the manifold using screws. And using GL45 as iron standard for a reactor solves a LOT of issues ;).

with-eight-hands, 2024-12-23:
Hmm. I'm not convinced that we need to keep the reagent tube diameter pretty large. Normally for a design like this, the reagent volume will be filling the tubing, and you would only start measuring after the line is primed, i.e. once the tube has been filled all the way up to the reactor entrance. If anything, tubing too large could result in space for air bubbles and uncontrolled flow. Larger tubing also means that we require more reagent solution just to operate the pump, which seems like a bad thing to me. The reagent in the line can potentially be pumped back into the syringe and recycled (although this is not good lab practice!), but it puts a lower bound on the amount of reagent you need to have regardless of how much you need for the reaction. As far as I know, no one has elaborated any target reactions for the microlab, but generally if we're thinking of reaction mixture volumes <200 mL, that's probably <10 g of reactants and products. So, it seems a bit silly to require twice that just to fill 30 cm of tubing. Also, 3 mm is about the largest flexible PTFE tubing that I could find.

Personally, I would not mess with wrapping a ring/disc with teflon tape. Teflon tape doesn't adhere to itself. If it is being clamped from all sides, it can form an OK seal, but the tension just from being wrapped won't prevent solvent vapors / splashes from working between the layers. I also imagine that the wrapping would become looser over time with loading and unloading. Maybe I am being a little obnoxious about the chemical compatibility stuff, but it seems like lots of people here haven't actually starting thinking about the chemistry part. I can't tell other people what to do, but I would absolutely try to avoid even incidental contact of heated solvents with random plastics in anything I might consume, even if it meant 20 hours of extra work. In this case, the effort to produce the lid that I made was probably <30 min, so to me it is an obvious choice.

A bit of an aside, but I was recently playing with cutting 8-mm teflon with a kitchen knife. So, you might not even need a hacksaw to approximate the teflon disc.

For the outer jar, I don't have any great solutions, but I do wonder about other attachment methods. I have thought about using french press coffee maker inserts. They're borosilicate, about the right size, and pretty available. I was wondering if a 3D print press-fit around the lip of a french press insert would be robust enough. It doesn't really need to seal or even sustain much force if the glass is supported from the bottom rather than suspended from the lid. Actually, one concept is how some french presses use a metal or plastic frame around the glass part, and the lid connects to this frame for a tight connection:

[image]

The print volume scares me a bit, but I guess it's not really larger than the 3D-printed stand.

I like your idea of modifying a metal lid, because it starts to resolve the issue of 3D print stability at high temperatures. I would love to replace the GL45 threaded part with something metal. Then, we're not limited to 80 C or whatever PETG can handle.

phaseloop, 2024-12-24:
Yeah, what you wrote about tubes and PTFE seal makes sense. I think it's good that you're vocal about chemical compatibility issues. I also suppose it will be possible in the future to design a 3d printed jig to help cutting PTFE seals.

While we are discussing french press glass - why not use a regular beaker? Then we can match expected reactor bottle volume to beaker volume to get close match and less thermal inertia (we're redesigning heating and power budget for a heater is pretty strict).

Technically we would not need any printed frame around it - only some kind of parametric fixture on top to match manifold to different beaker diameters and keep heater power lines in place.

Microlab v6 has the reactor hanging above table but I don't suppose this is necessary - it just needs to have thermal insulation and probably be placed inside some part to restrict movement or accidental toppling.

As for modifying a metal lid - I don't recommend that unless it's a last resort. Lids corrode awfully fast (at least jar lids).

GL45 thread could be printed on SLA printer in resin for high temperature performance (and/or printed in commercial facility on metal/nylon, etc) and then attached to the rest - printed cheaply in PLA/PETG.

with-eight-hands, 2024-12-29:
Oh, a 3D-printed jig for preparing PTFE is an excellent idea! A very simple thin template would be sufficient for most of the x-y placement, and a tube upward around the center hole location would help the stir rod hole be perpendicular without a drill press.

A beaker could make sense. I was thinking of French presses because some of them have the spout integrated with the lid, rather than as part of the glass. The lip of the glass vessel for this kind is a flat circle, which would work better with a press-fit reactor lid. But yes, if the spout is covered by part of the print or maybe even just with aluminum foil or something (to reduce evaporation and heat loss), beakers could work. As I understand it, they don't really have standard diameters for particular volumes, but that could be OK as long as people aren't afraid of parametric CAD to customize it.

Regarding my comment on metal, I was imagining the entire lid made of metal for temperature compatibility, since the entire outer vessel gets hot. I was focusing on the GL45 thread part because that is the hardest to make by hand. I realize this is getting a couple steps ahead of where things are now. I wonder if a design with clips like your Weck jars or standard taper Keck clips might be better than threads in the long run.

phaseloop, 2024-12-31:
All French presses, I've had or seen have a glass spout and technically are no different from a large beaker. We risk people chasing particular models of French presses. Another way is just using a glass jar but fitting the lid into internal hole diameter instead of using threads.

While whole metal lid would be nice - 3D printing it would be cost-prohibitive and difficult (external contracting). A middle-ground would be a heat-resistant GL45 thread (metal, resin) and the rest of the lid separated from hot elements (rubber gasket, etc). This should be an easy to add option as not all recipes call for high temperature that would damage the lid/thread.

sparklestheunicorn, 2025-01-07:
@with-eight-hands This is an amazing first crack at both the reactor hardening issue and the jar internationalization issue. Would love to get the FreeCAD files added to the parts repo as a Reactor Core variant when you think they're ready. It's OK if they're still "alpha" - we can add a README that explains the status and suggests directions for further development. Thanks for this thoughtful contribution to the project!