FourThievesVinegar/solderless-microlab #298: Move GPIO handling from python3-libgpiod to native gpdiod v2 python package

phaseloop, 2025-01-02:
python3-libgpiod system package was abandoned at version 1.5.4 and all development moved to native python package "gpiod".
Old package also seems to not be compatible with python docker images.

The issue is gpiod v2 package API is slightly different from the previous one and throws some errors on device initialization. This needs to be debugged and fixed. Then all references to python3-libgpiod should be removed and gpiod should be added to requirements.txt

mushkevych, 2025-05-11:
perhaps something like this will work:

  * rename `backend/hardware/gpiochip/gpiod.py` to `backend/hardware/gpiochip/gpiod_chip.py` in order to avoid naming collisions for the `gpiod`

```
from hardware.gpiochip.base import GPIOChip
import gpiod
from gpiod.line import LineSettings, Direction, Value
from util.logger import MultiprocessingLogger
from typing import Dict, List, Union, Any

LINE_REQ_DIR_OUT: str = 'output'
LINE_REQ_DIR_IN: str = 'input'

class GPIODChip(GPIOChip):
    def __init__(self, gpio_config: Dict[str, Any]) -> None:
        """
        Initializes the GPIO chip configuration.

        :param gpio_config: dict with keys:
            - chipName: device path for the gpiochip (e.g., '/dev/gpiochip0')
            - lineAliases: optional dict mapping alias strings to line offsets
        """
        self._logger = MultiprocessingLogger.get_logger(__name__)
        self.chip_name: str = gpio_config['chipName']
        self.line_aliases: Dict[str, int] = gpio_config.get('lineAliases', {})
        self.output_offsets: List[int] = []
        self.output_values: List[int] = []
        self._logger.debug(f"Configured chip {self.chip_name} with aliases {self.line_aliases}")

    def _get_line_number(self, pin: Union[str, int]) -> int:
        """
        Resolve a pin alias or return the numeric offset.
        """
        if isinstance(pin, str):
            if pin in self.line_aliases:
                return self.line_aliases[pin]
            raise ValueError(f"Invalid GPIO pin alias: {pin}")
        return pin

    def _commit_output(self) -> None:
        """
        Batch-request all configured lines as outputs and set their values using the modern API,
        opening and closing the chip device each time via context manager.
        """
        if not self.output_offsets:
            return

        config: Dict[int, LineSettings] = {}
        for off, val in zip(self.output_offsets, self.output_values):
            ls = LineSettings()
            ls.direction = Direction.OUTPUT
            ls.output_value = Value.ACTIVE if val else Value.INACTIVE
            config[off] = ls

        with gpiod.Chip(self.chip_name) as chip:
            with chip.request_lines(config=config, consumer="microlab") as req:
                # Initial values set by request_lines; nothing else required
                pass

    def setup(
        self,
        pin: Union[str, int],
        pinType: str = LINE_REQ_DIR_OUT,
        outputValue: int = 0
    ) -> None:
        """
        Configure a pin for output (only output supported).

        :param pin: alias or numeric line offset
        :param pinType: 'output' or 'input'
        :param outputValue: initial value (0 or 1)
        """
        line = self._get_line_number(pin)
        if pinType != LINE_REQ_DIR_OUT:
            raise NotImplementedError("Only output pins are supported")

        if line in self.output_offsets:
            idx = self.output_offsets.index(line)
            self.output_values[idx] = outputValue
        else:
            self.output_offsets.append(line)
            self.output_values.append(outputValue)

        self._commit_output()

    def output(
        self,
        pin: Union[str, int],
        value: int
    ) -> None:
        """
        Update the output value for a configured pin.

        :param pin: alias or numeric line offset
        :param value: 0 or 1
        """
        line = self._get_line_number(pin)
        if line not in self.output_offsets:
            raise ValueError(f"Pin {pin} has not been set up for output")

        idx = self.output_offsets.index(line)
        self.output_values[idx] = value
        self._commit_output()

```

0phelia36, 2025-05-19:
Completed in #323

closed 2025-05-19 by 0phelia36