FourThievesVinegar/solderless-microlab #302: As a user, I want a wide range of options for cooling the MicroLab

sparklestheunicorn, 2025-01-09:
Currently, the MicroLab can be cooled with an ice bath and a circulating fluid like water, acetone, or alcohol. We would like to have a wide range of options that can support a wide range of temperatures for long periods of time.

A collection of common cooling bath solvents and containers is here: https://chem.libretexts.org/Bookshelves/Organic_Chemistry/Organic_Chemistry_Lab_Techniques_(Nichols)/01%3A_General_Techniques/1.04%3A_Heating_and_Cooling_Methods/1.4J%3A_Cooling_Baths

We may also want to consider options that require less manual attention (restocking ice may be necessary for a DIY ice bath). A small cheap chest freezer is one potential solution: https://www.homedepot.com/pep/Magic-Chef-3-5-cu-ft-Manual-Defrost-Chest-Freezer-in-White-HMCF35W5/313922431

phaseloop, 2025-01-10:
Yeah, chest freezer or front-loaded "mini freezer" would probably be the best and cheapest solutions unless someone want to play DIY with compressors and plate heat exchangers (which won't be cheaper at all). Seems front loaded mini freezers are smaller than chest ones.

Depending on a model front-loaded mini freezers reach -22 to -14C as minimal temp.

sparklestheunicorn, 2025-10-20:
Docs have been updated to recommend some cooling solutions. Calling this closed.